indykoning/PyPi_GrowattServer #2: can't compile: tag = os.environ['LATEST_TAG'] ENV variable not set

onkelbeh, 2020-09-06:
Hi,

hope you don't mind that I use your integration in [Home Assistant Gentoo Overlay](https://github.com/onkelbeh/HomeAssistantRepository).

I am not using your component myself, but I maintain it for completeness.
Your can take a look of your components Ebuilds [here](https://github.com/onkelbeh/HomeAssistantRepository/tree/master/dev-python/growattServer).

Currently it is not possible to compile your component from scratch:
```
 * Package:    dev-python/growattServer-0.0.4
 * Repository: HomeAssistantRepository
 * Maintainer: b@edevau.net
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_targets_python3_7 test userland_GNU
 * FEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
>>> Unpacking growattServer-0.0.4.tar.gz to /var/tmp/portage/dev-python/growattServer-0.0.4/work
>>> Source unpacked in /var/tmp/portage/dev-python/growattServer-0.0.4/work
>>> Preparing source in /var/tmp/portage/dev-python/growattServer-0.0.4/work/growattServer-0.0.4 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/dev-python/growattServer-0.0.4/work/growattServer-0.0.4 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/dev-python/growattServer-0.0.4/work/growattServer-0.0.4 ...
 * python3_7: running distutils-r1_run_phase distutils-r1_python_compile
python3.7 setup.py build -j 10
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    tag = os.environ['LATEST_TAG']
  File "/usr/lib/python3.7/os.py", line 681, in __getitem__
    raise KeyError(key) from None
KeyError: 'LATEST_TAG'
 * ERROR: dev-python/growattServer-0.0.4::HomeAssistantRepository failed (compile phase):
 *   (no error message)
```
Home Assitant relies on `growattServer==0.0.4`
Problem still persists in current version

To fix this for nox, I added a small patch to my Ebuilds, where ${PV} is the current package version in the build environment:

```bash
src_prepare() {
	sed "s/tag = os.environ\['LATEST_TAG'\]/tag = '${PV}'/g" -i setup.py || die
	eapply_user
}
```
All works now, but you should get rid of this unusual query in setup.py.
Thanks a lot.

fabaff, 2021-05-15:
I would welcome the removal f the env var as well. Build issue for Fedora and NixOS.

indykoning, 2021-12-03:
I've moved over to using workflows to deploy new versions and in the process ensure i cannot forget to properly set the version
https://github.com/indykoning/PyPi_GrowattServer/blob/master/.github/workflows/release.yml#L16
I hope i will never need it :sweat_smile: but from now on the version tags should contain the latest tag without reading it from the environment.