indykoning/PyPi_GrowattServer #3: Error when using inverter_id

wynandmyriota, 2020-09-11:
I'm trying to get details from my inverter, but the inverter ID I'm using is giving me the following error message:

{'msg': '501', 'success': False}

I'm using the inverter ID that is displayed on the web page of the Growatt server:

test = api.inverter_data('NAC5A142E4', datetime.date.today())

indykoning, 2021-03-06:
Could you try using the deviceSn that shows up if you run:
`api.device_list(plant_id)`
Maybe the alias (which is what shows up on the website) differs from the inverter ID

orson1282, 2021-04-19:
I have the exact same issue.
I used the `deviceSn` that shows up when I run `api.device_list(plant_id)`
Also tried it with the `deviceAilas` and the `datalogSn`, I even removed the spaces I had in the alias, but the response is always 
`{'msg': '501', 'success': False}`

Any help would be appreciated.

indykoning, 2021-05-18:
Are you certain it is only an inverter and not a different device? e.g. mix/hybrid, battery
You can do so by checking the `deviceType`

yeeyangtee, 2021-12-18:
Hello! Encountering the same issue:
1. Get inverter ID with `api.device_list(plant_id)`, confirmed deviceType='inverter'
2. Using `deviceSn` from step one in `api.inverter_data` gives `{'msg': '501', 'success': False}`

indykoning, 2021-12-21:
Hmm after some digging it could maybe be a username/password issue as well 
https://github.com/DiedB/Homey-SolarPanels/issues/105#issuecomment-654303602

Could you maybe try changing your password and see if this fixes the issue? Maybe they changed their way of storing passwords after you created your account

iagomachadoeng, 2022-02-19:
> I'm trying to get details from my inverter, but the inverter ID I'm using is giving me the following error message:
> 
> {'msg': '501', 'success': False}
> 
> I'm using the inverter ID that is displayed on the web page of the Growatt server:
> 
> test = api.inverter_data('NAC5A142E4', datetime.date.today())

Same problem here, tested in 4 different inverters. I'd like to create a dashboard that integrate the day generation of each inverter in a month and report by email. Anyone got successful to get the daily generation of the month?