indykoning/PyPi_GrowattServer #13: 504 server response to mix_detail()

igorpolsh, 2021-05-18:
Hi, 

Im getting this: 
{"result":0,"obj":null,"msg":"504"}

as a response to  mix_detail().

called by api.mix_detail(PRB1234567, 10000) 

Is there anything I am missing or did the server side api change/was removed?

muppet3000, 2021-05-18:
504 is a weird error here, that implies a gateway timeout.
Can you try running the `mix_example.py`? It will prompt you for your credentials and then print out all of the information about your system(s).
That way we'll be able to rule out whether it's an issue with your implementation or the connection to the server. 
Let me know whether the mix example works and we can go from there.

igorpolsh, 2021-05-18:
I tried running it, but after the input of username it doesn't go any further. weird. 

Just to make sure the mix_id that we pass to the function, that is the inverter_id?

muppet3000, 2021-05-18:
I've just tried re-running it against my system and it still works for me. 

Does it just timeout immediately e.g. when it's trying to log in to the server? Or does it only time out when you're trying to call the `mix_detail` function?
At the very least the example script should output a list of "plants" registered to your account. If it's not doing that then there's definitely a connectivity problem.

With regards to your question about what should be passed to the function, I've copy/pasted/modified a bit of the example below to show what needs to be passed into the function:

```
#The login stuff
api = growattServer.GrowattApi()
login_response = api.login(username, user_pass)

#Get and print a list of plants
plant_list = api.plant_list(login_response['userId'])
print("***List of plants***")
for plant in plant_list['data']:
  indent_print("ID: %s, Name: %s"%(plant['plantId'], plant['plantName']), 2)
print("")

#Loops over each plant
for plant in plant_list['data']:
  plant_id = plant['plantId']
  plant_name = plant['plantName']
  plant_info=api.plant_info(plant_id)
  for device in plant_info['deviceList']:
    device_sn = device['deviceSn']
    device_type = device['deviceType']
    print("Device - SN: %s, Type: %s"%(device_sn, device_type))

  print("")
  for device in plant_info['deviceList']:
    device_sn = device['deviceSn']
    device_type = device['deviceType']
    #NOTE - This is the bit where we specifically only handle information on Mix devices - this won't work for non-mix devices

    mix_info = api.mix_info(device_sn, plant_id) #We pass the "deviceSn" as the `mix_id` and the "plantId" as the `plant_id` arguments
```
So it's not the inverter id, it's the value that is returned from the call for `plant_info`. 

If the `plant_info` doesn't respond correctly then it's more of an issue with your connectivity to the growatt API than the library. 

Have you tried signing in to https://server.growatt.com website in a web browser?

igorpolsh, 2021-05-18:
I am getting response from plant_info (devicesn seems to be also the inverterid) 

I am using http://server.smten.com/ - it is a clone growatt system. 
I have registered to growatt as well, and it gives the same error (but haven't set up my inverter with it at the moment because I am trying to get my history data off the smten server prior to switching).

Here is my code if it helps pinpoint the issue:
`import datetime
import growattServer
import sys
import getpass

def list_decode(data, criteria, decoded_criteria):
    do_something = False
    for list_item in data:
        if str(type(list_item)) == "<class 'list'>":
            decoded_criteria = list_decode(list_item, criteria, decoded_criteria)
        elif str(type(list_item)) == "<class 'dict'>":
            decoded_criteria = dict_decode(list_item, criteria, decoded_criteria)
        elif str(type(list_item)) == "<class 'tuple'>":
            decoded_criteria = tuple_decode(list_item, criteria, decoded_criteria)
        elif str(type(list_item)) == "<class 'str'>":
            if str(list_item).upper() == criteria.upper():
                decoded_criteria = str(list_item)
        elif str(type(list_item)) == "<class 'bool'>":
            do_something = True
    return decoded_criteria

def dict_decode(data, criteria, decoded_criteria):
    do_something = False
    for dict_item in data:
        if str(type(dict_item)) == "<class 'list'>":
            decoded_criteria = list_decode(dict_item, criteria, decoded_criteria)
        elif str(type(dict_item)) == "<class 'dict'>":
            decoded_criteria = dict_decode(dict_item, criteria, decoded_criteria)
        elif str(type(dict_item)) == "<class 'tuple'>":
            decoded_criteria = tuple_decode(dict_item, criteria, decoded_criteria)
        elif str(type(dict_item)) == "<class 'str'>":
            # print("STR: " + str(dict_item).upper() + " | criteria: " + criteria.upper())
            if str(dict_item).upper() == criteria.upper():
                decoded_criteria = str(data[dict_item])
        elif str(type(dict_item)) == "<class 'bool'>":
            do_something = True
    return decoded_criteria

def tuple_decode(data, criteria, decoded_criteria):
    do_something = False
    for index, tuple_item in enumerate(data):
        if str(type(tuple_item)) == "<class 'list'>":
            decoded_criteria = list_decode(tuple_item, criteria, decoded_criteria)
        elif str(type(tuple_item)) == "<class 'dict'>":
            decoded_criteria = dict_decode(tuple_item, criteria, decoded_criteria)
        elif str(type(tuple_item)) == "<class 'tuple'>":
            decoded_criteria = tuple_decode(tuple_item, criteria, decoded_criteria)
        elif str(type(tuple_item)) == "<class 'str'>":
            # print(str(tuple_item))
            if str(tuple_item).upper() == criteria.upper():
                decoded_criteria = str(tuple_item[index - 1])
        elif str(type(tuple_item)) == "<class 'bool'>":
            do_something = True
    return decoded_criteria

def decode_response(data, criteria):
    """
    This function acts a 'find' function in the data. designed to work with any data object returning from server

    decide handler function based on data type"""
    do_something = False
    decoded_criteria = "decode - error - criteria not found"
    returned_criteria = ""
    # print("Data type inside decode_response: " + str(type(data)))
[... 85 lines cut ...]
login_response = api.login(userId, userPass)
for response in login_response:
    if response == "success" and login_response[response] is True:
        print("Connected successfully")
    elif response == "success" and login_response[response] is False:
        print("Error connecting!")

printer = ""
data = api.plant_list(userId)  # get plant data via user id
plant_id = decode_response(data, "plantid")  
mix_id = "" 
inverter_id = "" 
data_logger = ""  

if plant_id != "error - criteria not found" and plant_id != "":  # check if plant id is found
    data = api.plant_info(plant_id)
    mix_id = decode_response(data, "deviceSn") # obtain data mix id

    data = api.device_list(plant_id)  # get device list
    inverter_id = decode_response(data, "deviceAilas")  # obtain inverter id
    data_logger = decode_response(data, "datalogSn")  # obtain data logger id

    data = api.mix_detail(mix_id, plant_id)
    printer = decode_response_format(data)
`

igorpolsh, 2021-05-18:
sorry, code block didn't really work there...

muppet3000, 2021-05-18:
Yeah that formatting really didn't go well!

I can't log in to the smten system because I don't have an account, even if I did, I would't have any systems registered there so wouldn't be able to debug. 

Are you sure that you have a "mix"/"hybrid" system? When you log in to the SMTEN website (assuming it's the same as the Growatt one), does it show that your system is a "hybrid" one (screen grab below from my setup):

[image]

If not then I can't guarantee that any of the `mix` function calls will work. (Although I wouldn't expect them to timeout, I would have expected an error code instead). 

What does your system get listed as when you're logged into the webiste?

igorpolsh, 2021-05-18:
[image]

igorpolsh, 2021-05-18:
if not mix_details, is there any other way of obtaining historical data but .5 hour increments?

muppet3000, 2021-05-18:
Ah, yeah, that would be why it doesn't work, you don't have a mix system. 

As for how to get the data, I don't have access to an "inverter" system, @indykoning implemented the inverter logic so he may be able to help out on that one. 

Looking at the similar function names in the library one of these ones may be able to get what you want: 
https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L90-L126
the `inverter_data` function is the one that takes a date as an argument, so that may be able to return chart data as well. 

Sorry I can't be more help, without access to an inverter system myself I wouldn't be able to implement that logic either.

muppet3000, 2021-05-18:
One option (which is a massive work-around) would be to download each of the CSV files day by day from the web url and then parse them programmatically, but that's a really ugly workaround. Would be much nicer to hit the API.

igorpolsh, 2021-05-18:
I'll be happy to give you my login details once i'm done getting the data off the server. Its purely historical data, ill remove all my account info from it and you can go for your life.

As I said though, after I get the data off it, I don't want to lose 2 years of data haha.

igorpolsh, 2021-05-18:
out of curiosity, have you had a look at the "Growatt Power station monitoring OpenAPI Protocol standards" 

its got a different api from what I can figure. there is just no record of how to login to it. strange. 

[Growatt-Server-Open-API-protocol-standards.pdf](https://github.com/indykoning/PyPi_GrowattServer/files/6500538/Growatt-Server-Open-API-protocol-standards.pdf)

muppet3000, 2021-05-18:
> I'll be happy to give you my login details once i'm done getting the data off the server. Its purely historical data, ill remove all my account info from it and you can go for your life.
> 
> As I said though, after I get the data off it, I don't want to lose 2 years of data haha.

I'd be happy to see what I can implement. I can't promise how quickly I'll be able to do it though as I only really maintain this stuff in my spare time. It shouldn't be too hard to implement though. 
You should be able to get my email address from my profile so you can email me the creds directly. 

I've just realised though that we've written this library by reverse engineering the android "shine" app, which points at the http://server.growatt.com url.
Do you have a mobile App that you use with the smten version?

I'll see what I can do though based on the work that @indykoning has already done in the library.

muppet3000, 2021-05-18:
> out of curiosity, have you had a look at the "Growatt Power station monitoring OpenAPI Protocol standards"
>
> its got a different api from what I can figure. there is just no record of how to login to [...]

Yeah I have seen this, however it seems like it was last published about 5 years ago and hits "v1" of the interface. I've always assumed that it was out of date and deprecated. Plus, I found this python library and thought "why would I bother writing another one when this one does (mostly) what I need".

igorpolsh, 2021-05-18:
> Do you have a mobile App that you use with the smten version?

I do, called smten. Pretty much the same thing. 
You think we can reverse engineer that and see how it gets the data? 

If so, have you got any material on how to do that? I'm new to python and json (ex java/php developer) but im sure ill work it out.

muppet3000, 2021-05-18:
Ahh, this one: https://baixarapk.gratis/en/app/1078137068/smten ??

So, the way that I've worked on it is to use an app called [NetCapture](https://play.google.com/store/apps/details?id=com.minhui.networkcapture&hl=en_GB&gl=US).

You set it up and let it run in the background while you use the SMTEN app (make sure you tell NetCapture to only listen to traffic for the app you're targeting otherwise you'll get every single transaction from your entire phone). Make sure you've gone to the relevant place in the app where you can get the data you're targeting (in your case it would be a graph of data). 

Then stop the capture in NetCapture and scroll through each of the transactions, each one will show the URL that was hit for the information and then the information that it returned. You need to find the one that returns the graph of data and then add a new function to the library that supports it (if it's not already hit by the library). 

Hopefully that's enough to nudge you in the right direction.

igorpolsh, 2021-05-18:
> Ahh, this one: https://baixarapk.gratis/en/app/1078137068/smten ??

that's the one! 
I'll give it a go, and get back to you with results

muppet3000, 2021-05-18:
Cool, if you get stuck and you're happy to share your creds with me just reach out and I'll have a look myself.

igorpolsh, 2021-05-18:
http://server-api.smten.com:80/newPlantDetailAPI.do?plantId=10039&date=2021-05-18&type=1 
should I be using this url for the api?

muppet3000, 2021-05-18:
Assuming that's what the NetCapture app spat out then yes.
The "http://server-api.smten.com" is what you should be using as the API. The `newPlantDetailAPI.do` is the endpoint you're hitting (that's what each of the functions in the python library target, each function hits a unique endpoint), then everything after the `?` is what is being passed as an argument to the endpoint.

igorpolsh, 2021-05-18:
calling the api via python gives me:
{'back': {'errCode': '002', 'success': False}}

muppet3000, 2021-05-18:
Have you added a new function to the `__init__` file in your local copy of this repository?
Doing a ctrl-f in here: https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py for `newPlantDetailAPI` shows that we don't currently have any implementation for it. 
It looks very similar to this function: https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L76-L88

You could basically just copy/paste that function and change the url on this line: https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L82 to have `new` at the beginning of it.

Remember that you need to have completed the "login" action in order for it to work.

igorpolsh, 2021-05-18:
I did add a new3 function to the __init__.py rep
I think its the same thing actually, just to clarify, what am I passing as 'timespan' on the function call? date? e.g 2021-05-18?

muppet3000, 2021-05-18:
Timespan is actually the value of an enum that is translated into the `type` part of the URL you posted above. 
The enum values are here: `https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L20-L23`
Whoever wrote the API at the growatt end did a really weird job, but if you want all the data from a given date you need to use `1` as the value for the timespan.

igorpolsh, 2021-05-18:
That is weird! I've noticed 'hour' or 0 gives me error 200 too. 
1 works, but thats without date. Can I pass date as string or does it have to be datetime type?

muppet3000, 2021-05-18:
If you don't give it a date it will default to your current local date.
If you want to pass a date in you need to pass in a python datetime object. The best way to do this if you're looking for a specific date's details is to do something like this:
```
from datetime import datetime
datestr = "2021-05-18" #18th May 2021
date_time_obj = datetime.strptime(datestr, '%Y-%m-%d')
```
Then pass the `date_time_obj` as the final argument to the function call.

igorpolsh, 2021-05-18:
ALRIGHT! It's working, giving me an hourly production too! 
I'll just loop this for the last 2 years and write the data out.

You're a legend! 

(sorry for the noob questions, I only been doing python for 2 days haha)

Do you still want/need my login details? Technically after I finish I will have enough to move on to growatt server and work it the way it was meant to with this rep. But if you still need the details (to play around with) - do let me know.

muppet3000, 2021-05-18:
No problem at all, happy to help. 

I don't need your login creds, but what would be really handy would be for you to submit a pull request with the new function call and updated documentation?

If this repo didn't already have the functionality to grab this sort of data from SMTEN then it's not going to be able to get it from growatt either. (Can you try calling the original function (the one without "new" in it) to see what data it returns in comparison to what you have from the new one).

If you're not comfortable submitting a pull request then copy/paste the exact code you've added here, ideally with an example of the data that's responded and when I get a chance I'll submit it myself, perhaps just getting you to validate it before it gets merged in.

Let me know either way.

igorpolsh, 2021-05-19:
Ok, so I've called the 'new...' function, and the old - seems to be exactly the same data. 

The only thing I have changed is the server url to 'http://server-api.smten.com' to make it work with SMTEN. 

Possibly the documentation could benefit noobs like me in the area where it describes the ENUM value and date formatting to be passed (e.g " call function 'Timespan.' followed by day/month/year" and date format should be in the following format '2020-05-18'") other than that, its spot on. 

also, since we figured it works on SMTEN, I think the new api url should go into the documentation too along with updated description on github to make users aware that this isn't just for growatt. 

I've never done any pull requests, so don't even know where to start, but I'm more than happy to help with more info

muppet3000, 2021-05-19:
Thanks for the feedback, glad to hear that the library doesn't need any further changes.

I agree with you comments on documentation. If I get the chance to make it a bit more noob-friendly soon I will. 

Don't worry about the PR, there's no need if the library already does what is needed. 

Are you happy if we close this ticket now?

closed 2021-05-19 by igorpolsh