indykoning/PyPi_GrowattServer #16: Library has stopped working

muppet3000, 2021-09-10:
As of 02:00 UTC the library has stopped working.

This has been tested by using the `mix_example` script

The following functions still appear to work:
`api.login`
`api.plant_list`

However, when the script attempts to call the `api.plant_info` function it returns the following response:
```
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.w3.org/TR/html4/loose.dtd">
<html>
<head>
<meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
<title>Error500</title>
</head>
<body>
        <br/>
        <br/>
        <br/>
                <center>
                <pre>
提示:尊敬的用户，您还未登录系统，请到登录界面进行登录 <br>
<a href="./login.do" target="_top">登录页面</a>
                </pre>

        </center>

</body>
</html>
```

I have attempted to cut-out this function call and then the next call in the script `api.mix_info` suffers the same problem.

I have validated using NetCapture that the API call still exists and should work (screen grab below)
[image]

I can't work out why the session is being dropped/ignored though. This must be a change on the server side, but I can't understand how the ShineApp still works but the library has stopped.

I have also tried changing the URL to the same one that the phone uses (https://server-api.growatt.com) however it makes no difference.

I'm not sure what to do next in terms of debugging and fixing as it looks like it's doing everything that it is supposed to.

TheWalrus2, 2021-09-10:
Same problem here. thnks for sharing info

TheWalrus2, 2021-09-10:
Other library https://github.com/johanmeijer/grott/issues/67
Also suffers some problems reading data format from growatt.

TheWalrus2, 2021-09-10:
tlx_deail call gives me a 500 response. So guessing that problem existst server-side.

TheWalrus2, 2021-09-10:
a got it working again.
change LoginApi.do in newLoginAPI.do
in the file _init.py_

muppet3000, 2021-09-10:
Looks like the response data is different, but yes, it works. I should be able to submit a PR in a bit (trying to juggle this in the background behind work).
Instead of:
```plant_list = api.plant_list(login_response['userId'])```
we now need:
```plant_list = api.plant_list(login_response['user']['id'])```
I'll also take a look at the HASS integration as well as discussed on the other thread.

Your suggestion, coupled with my change above makes the `mix_example.py` work again.

TheWalrus2, 2021-09-10:
pull request is made.  i don't use "mix_example" call.. but you changes look plausibele
 So 3 way merge and its fixed again :)

muppet3000, 2021-09-10:
Responded over there, if you can make the additional changes then it can all be in a single PR. We have to wait to do the PR for HASS until after the new release of the pypi library is made but I'll prep that tonight and then as soon as the release is made by Indy I'll get it approved and merged for HASS.

TheWalrus2, 2021-09-10:
Responded over there

indykoning, 2021-09-11:
Version 1.1.0 has been released now! This has the fix including the changes to keep breaking changes at bay :tada:

closed 2021-09-12 by muppet3000

soulraven, 2022-03-01:
The example and functions stop working again. none of the main functions are working.

muppet3000, 2022-03-03:
@soulraven I don't believe this library is at fault, nobody else has reported an issue and my system is working fine. 
Do you have more information about the context in which you're using this library?

soulraven, 2022-03-03:
i am in the process to document API from iOS app.
And all the url's are changed. Meaby some functions are working in legacy mode or something.

Now we have: newTwoLoginAPI.do, newTwoServiceAPI.do, newStorageAPI.do and some 2-3 new functions. the rest the functionally for the API is done using get parameters of the same function.
Now i am working to document the new API using OpenAPI.

will be soon on my repository growatt-monitor

[image]

muppet3000, 2022-03-03:
What type of system are you testing against? I'm looking at the mix_example and it seems to be working apart from some of the values on the `mix_totals` call.
What specific api calls/URLs are no longer working?

muppet3000, 2022-03-03:
Note - there is a comment on the Growatt Server website saying that the site may be down for maintenance on the 5th March, they may roll out a new API at that point and disable the old one (they've done it before) at which point we may need to move to the new calls that you've identified, but if that happens we'll need to do it reactively rather than pre-emptively change the library.

soulraven, 2022-03-03:
i am using Growatt 5000ES.

soulraven, 2022-03-03:
> Note - there is a comment on the Growatt Server website saying that the site may be down for maintenance on the 5th March, they may roll out a new API at that point and disable the old one (they've [...]

the iOS application was updated 3-4 days ago, or something like that. and the URL from the calls are new, nothing i from the old API, only the newLoginAPI.do seems to be old

muppet3000, 2022-03-03:
I'm not sure what "type" of system that is or what examples you're claiming from this library don't work. 
We'd happily welcome more contributions to this library if there is missing functionality.
Did you have a working version of this library before the update then? If so, what library calls no longer work? All of the ones in the `mix_example` definitely still work as I've just tested them.

soulraven, 2022-03-03:
the mix_example fails on mix_info, mix_totals, mix_detail. All of this functions fails and return null or empty json.

muppet3000, 2022-03-03:
Do you have a "mix" system then? Did it work before you made your original comment 2 days ago?
I have a mixed system myself and I've just run the example and it works as it always has done.

muppet3000, 2022-03-03:
My system is classed as a "mix" system which on the Growatt website is called "hybrid":

[image]

It would be worth checking what type of system yours is on the website before deciding what does/doesn't work. Not all of the API calls work for all types of systems.

soulraven, 2022-03-03:
no, only two pcs of Growatt 5000 ES. Only 2 days ago i have started to integrate the library in my own application, and started to test each method from the library. My system is off-grid classified as storage

[image]
[image]

muppet3000, 2022-03-04:
Ahhhhhhh that would explain it. This library covers all of the possible API calls for all types of systems, all of the calls aren't compatible with all types of system.
I'd recommend taking a look at the HomeAssistant implementation here: https://github.com/home-assistant/core/tree/dev/homeassistant/components/growatt_server for a really good implementation of how we choose what functions to call based on what type of system if being queried.
The best file to look at is the sensor.py file, specifically this if-else section: https://github.com/home-assistant/core/blob/dev/homeassistant/components/growatt_server/sensor.py#L187-L244

The home-assistant integration is really nice as it gives you all of the lovely energy monitoring capabilities. 

Either way, if you find new areas of the API for your storage/off-grid system please contribute to this library, we appreciate all of the updates and they feed directly into the HomeAssistant integration as well.

soulraven, 2022-03-04:
For the moment i will document the calls with openapi and them i will start to make a library for that api calls

muppet3000, 2022-03-04:
Ok, whatever works for you, it would be great to add those API calls to this library so everyone can benefit from them.

soulraven, 2022-03-04:
> Ok, whatever works for you, it would be great to add those API calls to this library so everyone can benefit from them.

be sure of that. 100% i will add to the library. but i consider a better documentation will be wonderful. Specially that the originals devs of the Growatt API have made some stupid mistakes.

Anyone has manage to get it working the original API? using the documentation from the growatt?

muppet3000, 2022-03-04:
Are you referring to this API? https://growatt.pl/wp-content/uploads/2020/01/Growatt-Server-API-Guide.pdf
That document was last updated in 2016, so I've always assumed that it's completely out of date. Hence why people like Indy and myself have added all of the functionality to this library.
I've done my best to document each of the values in the library itself e.g. https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L217-L247

soulraven, 2022-03-04:
> Are you referring to this API? https://growatt.pl/wp-content/uploads/2020/01/Growatt-Server-API-Guide.pdf That document was last updated in 2016, so I've always assumed that it's completely out of [...]

yes, that doc and that api.
I know about the documentation and is great, and will be included in the OpenAPI file that i writing right now. The more information the better.

soulraven, 2022-03-11:
i have made the first push of the swagger open api file from Growatt SPF 5000 ES inverters. For other types of inverters i don't know if will work.
https://github.com/soulraven/growatt-monitor/blob/main/docs/growatt-web-api-swagger.yml