indykoning/PyPi_GrowattServer #20: Sporadic missing value for currentpower

LuFo27, 2022-01-10:
The API sporadically shows 0 W currentPower despite the system running.
The correct value is displayed on the website.
I use the following calls. Does anyone else have the problem besides me?

login_response = api.login("xxxxx", "xxxxxxx")

plantdata = (api.plant_list(login_response['user']['id']))
print(str(plantdata))

{'data': [{'plantMoneyText': '3.8 (€)', 'plantName': 'xxxxx', 'plantId': 'xxxxxx', 'isHaveStorage': 'false', 'todayEnergy': '0.1 kWh',
 'totalEnergy': '12.8 kWh', 'currentPower': '0 W'}], 'totalData': {'currentPowerSum': '0 W', 'CO2Sum': '5.12 T',
 'isHaveStorage': 'false', 'eTotalMoneyText': '3.8 (€)', 'todayEnergySum': '0.1 kWh', 'totalEnergySum': '12.8 kWh'}, 'success': True}

Thanks and best regards

Derko01, 2022-03-08:
Ah, I just posted a similar problem. However, as of today it is not sporadic!

muppet3000, 2022-03-08:
Hi, sorry, I've got no idea how I missed this back in January. Looks like it's now been replicated in https://github.com/indykoning/PyPi_GrowattServer/issues/24 though. So let's continue the conversation in the most recent ticket.

muppet3000, 2022-03-09:
See #24 for 'fix' for this.
This ticket can be closed as a duplicate.

LuFo27, 2022-03-09:
Hello!
I adjusted my script as described in issue https://github.com/indykoning/PyPi_GrowattServer/issues/24
For the moment it works. Hopefully this will permanently fix my problem.
Thanks you all!

closed 2022-03-09 by LuFo27

muppet3000, 2022-03-09:
Fixed until Growatt decide to change something else on the API at least :)