indykoning/PyPi_GrowattServer #22: Growatt Open API

gilbertoe, 2022-01-26:
Hi
Could you please let me know if the GrowattServer is based the Growatt Open API?
Gilberto

soulraven, 2022-03-11:
i have made a reverse engineering from the Growatt iOS app. Is the first push to git.
Personally i have Growatt SPF 5000 ES inverters, for other types of inverters i am not sure will work.

https://github.com/soulraven/growatt-monitor/blob/main/docs/growatt-web-api-swagger.yml

Vic-s, 2023-04-29:
Moved to: https://github.com/soulraven/growatt-monitor/blob/main/docs/assets/growatt-web-api-swagger.yml

indykoning, 2025-05-11:
It was not based on Open API as it was not available (for consumers) yet at the time of creation.
However there is now work being done to deprecate the reverse-engineered version and move to OpenAPI
https://github.com/indykoning/PyPi_GrowattServer/discussions/107

closed 2025-05-11 by indykoning