indykoning/PyPi_GrowattServer #23: Login using a Installer Account

iagomachadoeng, 2022-03-05:
Anyone tried login using an Installer account? I'm a installer, when I try this code: 
"api.login(username, password)" using my credentials, I get this message: "{'success': False, 'msg': '502'}"

I wish I could get a list plant_list of all my instalattions.

muppet3000, 2022-03-06:
Hi @iagomachadoeng Not that I know of.
Do you use the same account to log in as an installer on the ShinePhone app?

It wouldn't surprise me if it went through a different set of API calls on the app for that. In order to look at it in more detail we'd need you to use something like the NetCapture SSL application to capture what calls the app was making. Alternatively, share your credentials with someone that could look at it for you (the latter requires a lot of trust of course).

muppet3000, 2022-03-09:
There's a very small bit of information at the bottom of the README for this library: https://github.com/indykoning/PyPi_GrowattServer#settings-discovery

Myself and Indy used the same technique. We installed the NetSSL app on our phones and then used that to capture traffic for the ShinePhone app only. You can then click through each transaction/packet that's sent. I reckon what happens is it fails the login for a "regular" user and then tries a different API call, but without having a set of credentials for an installer I wouldn't be able to do it myself. 
If you want to try yourself and report back your findings that would be awesome.
Good luck.