indykoning/PyPi_GrowattServer #30: Growatt API changed on June 14th?

JeroenHaringman, 2022-06-22:
Since June 14th 2022 16:25 CEST the module cannot log into the Growatt server, the login_response fails. For errors see this screenshot: https://www.dropbox.com/s/qf6q51e73vkvtu2/2022-06-22%2008_19_12-Window.png?dl=0

I upgraded to the latest version of the growattServer module (1.2.2) but that didn't solve this problem.

When googling the problem, I found more people who had problems with accessing the Growatt systems from June 14th onwards, so I'm assuming Growatt has tinkered with their API and told no one.

MarkTsengTW, 2022-06-22:
Hi, I think @muppet3000 took care of this one [a few days ago](https://github.com/indykoning/PyPi_GrowattServer/commit/0dce365d6e68c24e5cda6e06a831a01c46c2aa60). At least in my case, I was getting an error similar to yours and the fix worked. Maybe the new URL will work for you too.

JeroenHaringman, 2022-06-22:
If I search the entire memory card for "site-packages", I find only one location:

home\pi\.local\lib\python3.9\site-packages

However, there's nothing growatt-related in there. 

When I search the entire memory card for "growattServer" I find two folders:

usr\local\lib\python3.9\dist-packages\growattServer
usr\local\lib\python3.9\dist-packages\growattServer-1.2.2.dist-info

That first one seems to contain the actual module, as it has the __init__.py file. The second one has a couple of files I don't recognise, "DESCRIPTION.rst", "INSTALLER", "METADATA", and a couple of others.

In short, it looks like the dist-packages\growattServer is the only location where this module is installed.

(quick background, I know just enough about all of this to be dangerous, which means I don't know a lot but learn stuff as I go along)

muppet3000, 2022-06-22:
There's no reason you should be seeing this issue, you've listed the correct folders etc. Have you tried running your same script with growatt uninstalled? Does it still work then? If so you have it installed somewhere else.
The screengrab you've sent above is referring to your `/home/pi/.local` directory which I'm guessing is a local python env rather than the global one, it looks like that's where you need to update the version of the library.

JeroenHaringman, 2022-06-22:
"There's no reason you should be seeing this issue"

Well yeah, that what was I was thinking. Just to be like 101% sure, I pip uninstalled the module, tried to import it, and sure enough, error module not found,as expected. Pip installed the module, ran my script and it works. 😮

I do not know what happened that caused all this. I'm going to blame myself, maybe I made a typo earlier, or I didn't sacrifice my firstborn or whatever.

muppet3000, 2022-06-22:
Solar flare, that's my usual excuse. Glad it's resolved at least. Would you mind closing this ticket if you're happy this is now resolved?

JeroenHaringman, 2022-06-22:
Closing ticket, as this was probably my own fault, or _maybe_ a solar flare.

closed 2022-06-22 by JeroenHaringman