indykoning/PyPi_GrowattServer #32: Unable to Login with User credentials (502)

freneticforester, 2022-08-08:
Hello Guys,

When trying the example login in the readme i'm unable to login.
I've captured my network requests and when supplying the hash of the password that is send by the shinephone app,
I'm able to login.

However, when i use the hash_password method which is used in the login method. It doesn't work.

Sjord, 2022-08-09:
Thanks. Is the hash a little bit different or totally different? Passwords that contain punctuation sometimes don't end up correctly in a variable, because the shell interprets some characters. Is that possible for you? Did you try putting your password in single quotes?

Sjord, 2022-08-09:
Another possibility that I can think of is that your password contains a non-ASCII character, which is mapped differently to bytes in Java than in Python.

wcoekaer, 2022-08-18:
Actually they have changed the server model. there is server.growatt.com and server-us.growatt.com.
The python library in __init__.py has server_url = "https://server-api.growatt.com/' 

That doesn't work any more.

Changing that to server_url = 'https://server-us.growatt.com' works.
@freneticforester can you try as well? 

I had the same failure before error 502. After the change it works.

class GrowattApi:
    server_url = 'https://server-us.growatt.com/'
    #server_url = 'https://server-api.growatt.com/'

wcoekaer, 2022-08-21:
Ah never mind, I just noticed the api.server_url variable that can be set so no need to modify the init.py. I somehow missed that

indykoning, 2023-01-05:
Yeah they have indeed added more servers it seems, good to hear it's working now! I'll close this 🙂

closed 2023-01-05 by indykoning