indykoning/PyPi_GrowattServer #41: Retrieve battery percentage time-series data

christian-5-28, 2022-11-19:
Hi, 

First of all, thank you for the development and maintenance of this project, pretty useful and easy to use. I was wondering if it's possibile to retrieve the **time-series data related to the battery percentage level**, the same data used to display the battery chart in the mobile app (see screenshot attached below). 

I tried to use the api method "storage_energy_overview":

`json_data = api.storage_energy_overview(plant_id, storage_id=device_sn)` 

But the query fails with a returned value of type Nonetype. The api works well on my account for all the other methods so maybe the problem is the storage_id, but I could not find how to retrieve it so I used the "device_sn" variable. 

Hope you can help, thank you!

[image]

flopp999, 2023-10-24:
I also tried to get storage_id but I couldn't find how.
I also tried with all SN but not helping

mymatenige, 2023-12-31:
Sorry to add to the "me too", but has anyone stumbled across a solution to this?  Having the SOC over time would be very useful to me.