indykoning/PyPi_GrowattServer #44: Growatt API Documentation

TheDevick, 2022-12-10:
Hi. I'm planning to create an Lib, like yours, but in PHP, that consumes the WEB API from Growatt.

I was searching by a Growatt API documentation, and find a Strange Documentation [Here](https://growatt.pl/wp-content/uploads/2020/01/[Growatt-Server-API-Guide.pdf), but it seems to be not working. (Or I am consuming it wrong, maybe?)

After all, my question is: What documentation you use to build the project?

muppet3000, 2022-12-11:
Welcome to the fun world of trying to reverse engineer the way the growatt APIs!
We've repeatedly tried to engage with Growatt on up-to-date documentation but they never respond.

Someone has already made an attempted at documenting the web api here: https://github.com/soulraven/growatt-monitor (look under the docs folder), there was a swagger file in there at some point but I have no idea what happened to that.

If you take a look at our readme here: https://github.com/indykoning/PyPi_GrowattServer#settings-discovery it gives an explanation of how we've reverse engineered the API from the Android app.

TheDevick, 2022-12-11:
Great! I'll take a look
Thanks.