indykoning/PyPi_GrowattServer #45: Error 403 Blacklist

gconceicao, 2022-12-19:
I am trying to login with a lot of users and with others internet connetion wifi, but I am receving this errors.

The response content of the login is the issue below. 

Then the software is broken because did not received the correct answers.

C:\Users\gconceicao\OneDrive - WEG EQUIPAMENTOS ELETRICOS S.A\Documentos\GitHub\Report-Viewer\venv\User_Plant_Info.py:55: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
C:\Users\gconceicao\OneDrive - WEG EQUIPAMENTOS ELETRICOS S.A\Documentos\GitHub\Report-Viewer\venv\Lib\site-packages\urllib3\connectionpool.py:1043: InsecureRequestWarning: Unverified HTTPS request is being made to host 'server-api.growatt.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
b'<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN"><html><head><title>403 Forbidden</title></head><body><h1>403 Forbidden</h1><p>You don\'t have permission to access the URL on this server.</p><p>denied by UA ACL = blacklist</p><hr/>Powered by Tengine <br/>CDN Request Id: a3b53e9516714791325488312e</body></html>'
C:\Users\gconceicao\OneDrive - WEG EQUIPAMENTOS ELETRICOS S.A\Documentos\GitHub\Report-Viewer\venv\Lib\site-packages\urllib3\connectionpool.py:1043: InsecureRequestWarning: Unverified HTTPS request is being made to host 'server-api.growatt.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
C:\Users\gconceicao\OneDrive - WEG EQUIPAMENTOS ELETRICOS S.A\Documentos\GitHub\Report-Viewer\venv\Lib\site-packages\urllib3\connectionpool.py:1043: InsecureRequestWarning: Unverified HTTPS request is being made to host 'server-api.growatt.com'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(

muppet3000, 2022-12-19:
Are you trying to use this software on a corporate network by any chance? I'm guessing you work for "WEG EQUIPAMENTOS ELETRICOS S.A" from the looks of your OneDrive path?

As per the error message you're receiving: 
```
You don't have permission to access the URL on this server.

denied by UA ACL = blacklist
```
It seems pretty self explanatory to me, you're being blocked from accessing the site by your corporate firewall. Not a problem with this library at all from what I can tell.

As for the certificate error, this can sometimes happen with corporate firewalls when they redirect you to a holding page rather than the intended site.

Good luck getting it resolved.

Sjord, 2022-12-20:
This seems more like Growatt is blocking the user agent. The error message "denied by UA ACL = blacklist" hints towards the user agent blacklist. This is a known problem (#38, #36). You could try updating your PyPi_GrowattServer to the latest version; some changes to the user agent were recently made (#42).

indykoning, 2023-01-05:
Could you check if it still happens from https://github.com/indykoning/PyPi_GrowattServer/releases/tag/1.3.0 onwards? 
You may have to set https://github.com/indykoning/PyPi_GrowattServer#initialisation up

muppet3000, 2023-01-05:
Just a bit of feedback on this now the latest releases are out there.

1. The build-in agent-identifier has been blocked again which causes a 403 error like this
2. I've also managed to get a 403 error when using this library if I absolutely hammer the API i.e. multiple big requests in less than 10 seconds - they instantly blacklist your IP

@gconceicao - Like Indy has suggested, I would try using the `agent_identifier` option that's part of the initialisation of the library to see if you can make this work. Also make sure you're not making too many requests or you'll get blocked.

gconceicao, 2023-01-06:
> This seems more like Growatt is blocking the user agent. The error message "denied by UA ACL = blacklist" hints towards the user agent blacklist. This is a known problem (#38, #36). You could try [...]

Thanlk you, that is it.

gconceicao, 2023-01-06:
> Just a bit of feedback on this now the latest releases are out there.
>
> 1. The build-in agent-identifier has been blocked again which causes a 403 error like this
> 2. I've also managed to get a 403 [...]

I updated the GROWATT lib to the oldest version, and change the agent as same as shine app with more 6 randow ASCII in the final of the user Agent, and it is work properly now.

Thnk you all for the tips.

marcovtwout, 2024-02-14:
Fixed by https://github.com/indykoning/PyPi_GrowattServer/pull/77
@indykoning this issue could be closed