indykoning/PyPi_GrowattServer #46: inverter_data() function

tyrail311, 2022-12-21:
Hello there,

I am trying to make a call to the inverter_data function with the following:
    pprint(gw.inverter_data(inverter_sns[0], date=datetime.datetime(2022, 12, 20).date()))

gw is the api object and the inverter_sns is a list of the serial numbers for each inverter. I tried to call the date with a string 
    pprint(gw.inverter_data(inverter_sns[1], date='2022-12-20'))and received the following error: 

        AttributeError: 'str' object has no attribute 'strftime' 

when I tried call above with the datetime object, and got the following response:

    {'msg': '501', 'success': False}

Is the format of my request wrong with the date, or is there an underlying issue with the api?

I tried to copy the code and run it locally, it seems like there is something wrong with the code when I run it locally with the JSON decoder at the following function call: 
(link to git repo)[https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#:~:text=def%20inverter_data(self%2C%20inverter_id%2C%20date%3DNone)%3A] 
Error:
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 3 column 1 (char 2)

martinqs85, 2024-06-15:
i have the same issue, did you manage to solve it?

CoenHordijk, 2024-06-15:
> i have the same issue, did you manage to solve it?

Hi, because of the issues that kept popping up with growattServer I set up a RS485 interface with my Growatt