indykoning/PyPi_GrowattServer #48: Dont work from now?

Grandicek, 2022-12-27:
Hello,
i use this project for months, but for now dont work.
Maybe growatt update? :/

muppet3000, 2022-12-27:
The user-agent has been blocked by Growatt (again).
Take a look at the documentation and the User Agent Options example file here: https://github.com/indykoning/PyPi_GrowattServer/blob/master/examples/user_agent_options.py

Grandicek, 2022-12-27:
i hate them, rly.....

MrLoskarn, 2022-12-28:
Yes, the link above fixed the problem. Thanks to muppet3000 for his hint. 
I tried it and changed my code to: 
api = growattServer.GrowattApi(False, "my-user-id")     # Use something usefull for my-user-id like "Plant_XYZ" 
and then it immediatly worked again.

Grandicek, 2022-12-28:
> Yes, the link above fixed the problem. Thanks to muppet3000 for his hint. I tried it and changed my code to: api = growattServer.GrowattApi(False, "my-user-id") # Use something usefull for my-user-id like "Plant_XYZ" and then it immediatly worked again.

hohoho, thank you :)

closed 2022-12-28 by Grandicek

kineci, 2023-09-23:
> Yes, the link above fixed the problem. Thanks to muppet3000 for his hint. I tried it and changed my code to: api = growattServer.GrowattApi(False, "my-user-id") # Use something usefull for my-user-id like "Plant_XYZ" and then it immediatly worked again.

Can you please explain where to change this? I'm having the same issue