indykoning/PyPi_GrowattServer #49: Issue with api

lcsztl, 2022-12-29:
It was working good but I start to receive this message instead

Traceback (most recent call last):
  File "C:\Users\lucas\PycharmProjects\teste\main.py", line 13, in <module>
    login_response = api.login(gw_username, gw_user_pass)
  File "C:\Users\lucas\PycharmProjects\teste\venv\lib\site-packages\growattServer\__init__.py", line 132, in login
    data = json.loads(response.content.decode('utf-8'))['back']
  File "C:\Users\lucas\AppData\Local\Programs\Python\Python310\lib\json\__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "C:\Users\lucas\AppData\Local\Programs\Python\Python310\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Users\lucas\AppData\Local\Programs\Python\Python310\lib\json\decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Sjord, 2022-12-29:
See also [Dont work from now? · Issue #48 · indykoning/PyPi\_GrowattServer](https://github.com/indykoning/PyPi_GrowattServer/issues/48)

mvf-ss, 2023-01-01:
It might be worth failing gracefully - with an error message including the response - when the login fails. (I'm not a python programmer and it's just taken me a couple of hours of stepping through a .py file with pdb to find out why the code is dying with a JSON decode error. Takes me back to debugging in the '70s on a teletype terminal.) I guess I could/should have come here first. Maybe an error message could even include the suggestion to visit this site if the response is 403.

indykoning, 2023-01-05:
This issue seems to be caused by Growatt returning an unexpected http code and this package not handling that correctly.
It may have to do with the user agent being needed to be randomised
https://github.com/indykoning/PyPi_GrowattServer#initialisation

I do not believe this package should silently handle these errors since that can cause unexpected behavior in applications implementing this package, however the thrown exception can be clearer and must be easier to be properly caught and debugged.
This package will throw proper http exceptions in case the response code is not what is expected from https://github.com/indykoning/PyPi_GrowattServer/releases/tag/1.3.1 on 🙂 

as an example
```
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://server-api.growatt.com/newTwoLoginAPI.do
```

closed 2023-01-05 by indykoning