indykoning/PyPi_GrowattServer #53: Discharge Stopped Soc

fisken, 2023-01-10:
I've read through the documentation and is it correct that you can't set the "Discharge Stopped Soc" with this package?

I'm making a program that decides whether to use power from the battery. And the only way is to set the "Discharge Stopped Soc" in the settings. I hope to use this package, as I use it to determine whether to charge the batteries.

muppet3000, 2023-01-10:
You should be able to do that, however you'd need to work out what parameters to set via the settings call. You'd need to monitor the calls that are made via the App and then replicate them with the API

fisken, 2023-01-12:
Thanks for the quick reply. Haven't tried monitoring calls before, but I gave it a try.
The app can't set "discharge stopped Soc" but I figured out I could use the "grid first" setting where I discharge the battery. Then I set the discharge rate to 0, at I now only user power from the grid, great for these winter times when power is cheap at night.

I used NetCapture to figure out the command and parameters it expects. It was more straightforward than I thought. The command to set this is "mix_ac_discharge_time_period", and here is an example with parameters for others to use:

`#Set inverter schedule for not charing
schedule_settings = ["0", #Discharging power %
                     "10", #Stop Discharging SoC %
                     "23", "05", #Schedule 1 - Start time
                     "03", "00", #Schedule 1 - End time
                     "1",        #Schedule 1 - Enabled/Disabled (1 = Enabled)
                     "11", "00", #Schedule 2 - Start time
                     "15", "00", #Schedule 2 - End time
                     "1",        #Schedule 2 - Enabled/Disabled (0 = Disabled)
                     "00", "00", #Schedule 3 - Start time
                     "00", "00", #Schedule 3 - End time
                     "0"]        #Schedule 3 - Enabled/Disabled (0 = Disabled)
print("Setting the inverter discharging schedule to:")
pp.pprint(schedule_settings)
response = api.update_mix_inverter_setting(device_sn, 'mix_ac_discharge_time_period', schedule_settings)
print(response)`

Closing the thread.

closed 2023-01-12 by fisken

muppet3000, 2023-01-12:
Thanks @fisken glad you got it working. Maybe it's worth submitting a PR to the repo with an example, or a new entry to the README.md file with a new bullet point under this section: 
https://github.com/indykoning/PyPi_GrowattServer#inverter-settings

That way other people can benefit from this knowledge in the future?