indykoning/PyPi_GrowattServer #54: api.tlx_detail(tlx_id) not getting all timestamps

codac, 2023-01-17:
Calling the TLX Detail API with the following command results in a restricted ouput.
Call:
`api.tlx_data(device_sn, date=dayToProcess)`
The values of the key "invPacData" results always with a start at 11:15 and end at 17:10 even though there are data points before 11:15 and after 17:10. No matter what day you provide, its always the same. Every datapoint before and after gets ignored.

```
    'invPacData': {   '2022-11-06 11:15': 386,
                      '2022-11-06 11:30': 623.2000122070312,
                      '2022-11-06 11:35': 623.7000122070312,
                      '2022-11-06 11:40': 624.2999877929688,
                      '2022-11-06 11:45': 622.7000122070312,
                      '2022-11-06 11:50': 621.7666625976562,
                      '2022-11-06 11:55': 619.933349609375,
                      '2022-11-06 12:00': 621.933349609375,
                      '2022-11-06 12:05': 622.2666625976562,
                      '2022-11-06 12:10': 622.6000366210938,
                      '2022-11-06 12:15': 621.7000122070312,
                      '2022-11-06 12:20': 620.8500366210938,
                      '2022-11-06 12:25': 621.7000122070312,
                      '2022-11-06 12:30': 621.7333374023438,
                      '2022-11-06 12:35': 622.25,
                      '2022-11-06 12:40': 622.5499267578125,
                      '2022-11-06 12:45': 627.6500244140625,
                      '2022-11-06 12:50': 627.4000244140625,
                      '2022-11-06 12:55': 629.7333374023438,
                      '2022-11-06 13:00': 628.5999755859375,
                      '2022-11-06 13:05': 365.3999938964844,
                      '2022-11-06 13:10': 144.3000030517578,
                      '2022-11-06 13:15': 142.39999389648438,
                      '2022-11-06 13:20': 170.60000610351562,
                      '2022-11-06 13:25': 149.5,
                      '2022-11-06 13:30': 154,
                      '2022-11-06 13:35': 121.76666259765625,
                      '2022-11-06 13:40': 159,
                      '2022-11-06 13:45': 153.60000610351562,
                      '2022-11-06 13:50': 144.43333435058594,
                      '2022-11-06 13:55': 142.59999084472656,
                      '2022-11-06 14:00': 145.03334045410156,
                      '2022-11-06 14:05': 154.5,
                      '2022-11-06 14:10': 133.45001220703125,
                      '2022-11-06 14:15': 91.70000457763672,
                      '2022-11-06 14:20': 81.85000610351562,
                      '2022-11-06 14:25': 71.19999694824219,
                      '2022-11-06 14:30': 58.39999771118164,
                      '2022-11-06 14:55': 56.633331298828125,
                      '2022-11-06 15:00': 57.39999771118164,
                      '2022-11-06 15:05': 57.133331298828125,
                      '2022-11-06 15:10': 57.133331298828125,
                      '2022-11-06 15:15': 57.83333206176758,
                      '2022-11-06 15:20': 56.13333511352539,
                      '2022-11-06 15:25': 53.16666793823242,
                      '2022-11-06 15:30': 50.30000305175781,
                      '2022-11-06 15:35': 39.70000076293945,
                      '2022-11-06 15:40': 33.5,
                      '2022-11-06 15:45': 31.96666717529297,
                      '2022-11-06 15:50': 35.900001525878906,
                      '2022-11-06 15:55': 34.29999923706055,
                      '2022-11-06 16:00': 35.25,
                      '2022-11-06 16:05': 30.5,
                      '2022-11-06 16:10': 22.900001525878906,
                      '2022-11-06 16:15': 17.25,
                      '2022-11-06 16:20': 13.600001335144043,
                      '2022-11-06 16:25': 12.533332824707031,
                      '2022-11-06 16:30': 11.833333015441895,
                      '2022-11-06 16:35': 10.766666412353516,
                      '2022-11-06 16:40': 9.566666603088379,
                      '2022-11-06 16:45': 8.733333587646484,
                      '2022-11-06 16:50': 8.300000190734863,
                      '2022-11-06 16:55': 8.233333587646484,
                      '2022-11-06 17:00': 8.066666603088379,
                      '2022-11-06 17:05': 8.050000190734863,
                      '2022-11-06 17:10': 7.550000190734863},

```

Further more the format/values of the numbers is different. For example:
'2022-11-06 11:15': 386,     #comma but no decimals; but in the growatt ap there is a decimal of .25
'2022-11-06 11:30': 623.2000122070312,     #here you have the proper decimals

muppet3000, 2023-01-17:
I'd have to take a look at this myself, I bet the app has started making a different API call. 
I'm planning on doing some investigations with the library over the coming weeks to add new functionality for the Home Assistant plugin, so I could take a look at it then.
It's certainly odd behaviour. 
Reminder - all this library does it pull data from the API/Servers, it doesn't manipulate it at all, what that probably means is that the Android App is either hitting a different endpoint or adding extra parameters to get the full day of data.

codac, 2023-01-25:
Alright @muppet3000 , that would be a big help, thank you.

codac, 2023-04-06:
@muppet3000 any update on this?

Vic-s, 2023-04-28:
@codac regarding your difference in data between the API and the app. I _think_ the API data is the power generated by the solarpanels (DC) and the app show the power delivered to your house (AC@230V). If you look at your account on the Growatt website you will see:

**dashboard**: hover over the graph, it shows 'solar' + power
**energy**: (select PAC) site + power
**energy**: (select PV1, PV2)  PV1 + power, PV2 + power

Now notice 'energy PAC' is always less then 'energy PV1 + PV2' and 'energy PV1 + PV2' is equal to dashboard and API power.

However the API data is labelled 'invPacData', inverter Power AC data (I guess). This is confusing.

From the website (HTTP response):
**dashboard**:  charts object with data labelled 'ppv'
**energy**: (select PAS) datas object with data labelled 'pac'
**energy**: (select PV1)  datas object with data labelled 'ppv1' and 'ppv2'

Vic-s, 2023-04-28:
@codac regarding missing data. I can not repoduce this. I do notice your data is chronological ordered. The data I receive is in random order. But it might you have ordered the data to make it more readable.
`api.tlx_data(device_serial_number,datetime.datetime.now() - datetime.timedelta(days=2))['invPacData']

{'2023-04-27 16:20': 1722.5501708984375,
 '2023-04-27 18:45': 734.5,
 '2023-04-27 15:50': 1869.7667236328125,
....
'2023-04-27 20:05': 79.4000015258789}`

And chronological
`data = api.tlx_data(device_serial_number,datetime.datetime.now() - datetime.timedelta(days=2))['invPacData']
print( [{k:data[k]}  for k in sorted( dict(ChainMap(*[data])))])

[{'2023-04-27 06:10': 0},
 {'2023-04-27 06:15': 0},
 {'2023-04-27 06:20': 0},
 {'2023-04-27 06:25': 20.399999618530273},
...
 {'2023-04-27 21:15': 1.2999999523162842}]
`

codac, 2023-05-19:
@Vic-s Thank you for your reply. 
I have not ordered the `api.tlx_data(device_sn, date=dayToProcess)` call. 
This still doesn't really answer my questions, doesn't it?
The time restriction between 11am and 5 pm as well as the missing decimals, I don't get how this is in any concern with your answers. Maybe it's because of my bad english skills, sorry for that...