indykoning/PyPi_GrowattServer #55: Growatt blocking accounts

TheWalrus2, 2023-02-07:
I think this the end of the Api. Growatt is locking accounts. Mine has been locked. API not working and shinapp says account locked.

muppet3000, 2023-02-07:
Oh the positivity :P

I recommend taking a look over here: https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15

I recommend reading my specific comment here: https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1421417491

Use the FAQ I put together over there to set up an alternative account via the website. 

Furthermore, if you switch back to using the "server.growatt.com" url (which seems to work again) it's works with the same credentials.

My original account still seems to be blocked on my phone but works fine on the website so I created myself a new login just for my phone and also one for my various automations so that one can't break another etc.

muppet3000, 2023-02-08:
Note - my account was blocked on my phone yesterday and is unblocked again this morning.

TheWalrus2, 2023-02-08:
Mine was unlocked this morning. But is locked again. As is said this could be the end of the API. Because Growatt is now blocking the accounts activly.

muppet3000, 2023-02-08:
> Mine was unlocked this morning. But is locked again. As is said this could be the end of the API. Because Growatt is now blocking the accounts activly.

Only on the shinephone app, not on the website and you can just change the endpoint to use the website again. 
Did you read my message from yesterday?

TheWalrus2, 2023-02-08:
I don't you understand the whole picture here. They activly Blocking accounts. So the API has become useluss. Website is not the problem.

muppet3000, 2023-02-08:
I understand fine. 
Just change the endpoint URL, they provide exactly the same API endpoints.

TheWalrus2, 2023-02-08:
Oke but that does not change the blocking of accounts problem...

TheWalrus2, 2023-02-08:
Website is also off air now.

muppet3000, 2023-02-08:
Everything has been very intermittent for the last 2 days, I suggest we just wait it out to see what changes they've got in store this time :P

Snux, 2023-02-08:
I had to change the API endpoint to the original server.growatt.com as could never get the new one to work.  Not had any account locking issues.  For my use case (a Pi with a small display showing some stats) I only poll the API every 5 minutes, maybe they're stopping calls that are more aggressive.  Let's see what happens over the next little while.

muppet3000, 2023-02-08:
I reduced my polling to every 3 minutes earlier today and things stabilised (however the data that's available on the server itself has been very flaky all day, but that's a server problem).

Snux, 2023-02-08:
I certainly had some rubbish figures on their normal website yesterday.  All the "now" info was fine, but the "today so far" was all out.

TheWalrus2, 2023-02-08:
I created extra accounts for app and API reader. Put the API reader on a 3 minute interval. Hopefully this will do the trick.

Unwies, 2023-02-08:
> I reduced my polling to every 3 minutes earlier today and things stabilised (however the data that's available on the server itself has been very flaky all day, but that's a server problem).

Can you please tell me where I change the polling time. thx

muppet3000, 2023-02-08:
> > I reduced my polling to every 3 minutes earlier today and things stabilised (however the data that's available on the server itself has been very flaky all day, but that's a server problem).
> 
> Can you please tell me where I change the polling time. thx

It depends how you're using this library, it's entirely dependent on how often whatever code you're using makes a call to this library.

muppet3000, 2023-02-08:
> > > > I reduced my polling to every 3 minutes earlier today and things stabilised (however the data that's available on the server itself has been very flaky all day, but that's a server problem).
> > > 
> > > 
> > > Can you please tell me where I change the polling time. thx
> > 
> > It depends how you're using this library, it's entirely dependent on how often whatever code you're using makes a call to this library.
> 
> Sorry, I'm not that deep into this. I'm using Home Assistant 2023.2.3 on a Raspberry PI 3. 

In which case you'll be using the stock Growatt integration and that is currently fixed at every 5 minutes.
I'd be VERY surprised if you've had your account locked with a 5 minute polling interval. Are you sure you've been blocked? What happens when you use the same credentials to log into the ShinePhone app? Do you get a pop up saying "this account is locked"?

muppet3000, 2023-02-08:
Ok, so you're using the HACS version, which also defaults to 5 minutes unless you've followed my FAQ on how to increase the polling interval.
In which case I recommend reaching out in the issues section over there rather than on this library, there is a chat going on on this ticket:

Derko01, 2023-02-10:
Hm, my ShinePhone-APP says: `The current account is locked.`
And my Python-code says: `plant_list = api.plant_list(login_response['user']['id'])     KeyError: 'user'`
And that is whether I use server.growatt.com of server-api.growatt.com
I am not happy.

muppet3000, 2023-02-10:
I recommend taking a read over here: https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15 where there's a very detailed discussion about it.
Specifically these two comments:
https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1423787751
&
https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1424810061

Note - when you poll at around 5 minutes your account doesn't seem to get blocked as quickly. Any more frequently than that and you get locked out.

As I've said in those comments above, you can create yourself another account through the web UI without any problem and use that. The lock seems to last approximately 24 hours.

TheWalrus2, 2023-02-10:
Join the club. We all had that issue. Mine is good again. Interval 5 minutes and new account.

Derko01, 2023-02-10:
OK, I had my interval set on 120s.
I just changed that to 300s.
I will wait 24 hours before activating my script again. Hopefully I am unlocked by then.
Thanks!

Derko01, 2023-02-10:
> > OK, I had my interval set on 120s. I just changed that to 300s. I will wait 24 hours before activating my script again. Hopefully I am unlocked by then. Thanks!
> 
> You can create a new sub-account under your existing one, follow the steps here: https://github.com/muppet3000/homeassistant-growatt_server_api/blob/dev/FAQ.md#how-do-i-share-my-credentials-with-you-for-testingadding-new-features-for-my-system-type

Great. Created an alternative account and it worked again with that! Thank yo very much!

DutchKillerbee, 2023-02-28:
> You can create a new sub-account under your existing one, follow the steps here: [...]

Stupid question: Where can I change the polling interval?
I am using your HACS version [homeassistant-growatt_server_api](https://github.com/muppet3000/homeassistant-growatt_server_api)

muppet3000, 2023-02-28:
> Stupid question: Where can I change the polling interval?
> I am using your HACS version [homeassistant-growatt_server_api](https://github.com/muppet3000/homeassistant-growatt_server_api)

Unless you've changed it from the default using this how to guide: https://github.com/muppet3000/homeassistant-growatt_server_api/blob/dev/FAQ.md#how-do-i-increase-the-update-interval-to-be-more-than-every-5-minutes

then you'll be using the default of 5 minutes.

DutchKillerbee, 2023-03-01:
Yes I am using it. But unfortunately my main growat user account is locked. 
I tried to create a new one but I don't receive any email.

So I checked your source and found in sensor.py at line 35 this:

`SCAN_INTERVAL = datetime.timedelta(minutes=5)`

I changed it to

`SCAN_INTERVAL = datetime.timedelta(minutes=10)`

Will this change the polling?
Currently I  am waiting until my account is unlocked.
Fingers crossed.......

muppet3000, 2023-03-01:
So you can't even log into the server.growatt.com website? That's the first I've heard of that happening.

You can change the code like that if you want, but 5 minutes should be enough.

DutchKillerbee, 2023-03-01:
With 5 min polling my account/ip was blocked. 
I received a message on my phone using the growatt app something like " account blocked". I only get that message once.
After that messages like "network time out"
All using Wifi at home.

I disabled Wifi at my mobile phone and I could login again using 4G. But after one hour not anymore....
Not from my pc, nor from my mobile phone (using Wifi or 4G). So it seems they are blocking IP addresses as well.....

Today I rebooted HA and lucky me I was able to get data again :) 
Polling is now at 10 minutes... Fingers crossed.

Maybe a suggestion:
Maybe we can disable polling between sunset and sunrise?

muppet3000, 2023-03-01:
That's a conversation for over in the actual repo for the Integration, not this repo.

Also, my answer is probably going to be a 'no' because that's quite a bit of effort to implement and I'm encouraging people to switch to Grott anyway and I'm in the process of making a new HA integration for that instead.

With regards to your account getting blocked:
1 - Yes, they block your IP, I just restart my router and I get a new IP when that happens

2 - They don't block your account on the website, only for the mobile app, therefore you can create yourself multiple accounts (I have 4 now for HA and a separate one for my phone) which you can toggle between to work around the problem.
There's an FAQ for how to do this on the integration repo (just scroll down further on the page I linked to yesterday and it'll show you how to do this).

Ultimately the use of the API is becoming less and less viable, hence my recommendation for people to try and switch to Grott, which isn't the most 'user friendly' experience, however, once I've got something up and running nicely for myself I'm hoping to contribute back to the main repo to make it more user friendly for people.

DutchKillerbee, 2023-03-01:
Thanks for your feedback.

staticweb, 2023-03-12:
Has anyone been in contact with growatt about this issue ?? if, what was their response ?

muppet3000, 2023-03-13:
> Has anyone been in contact with growatt about this issue ?? if, what was their response ?

I have tried repeatedly to contact growatt (for a variety of things related to supporting the API), I have tried every form of email address I can find for them. I have received zero responses from them.

Ultimately, while I don't like what they're doing, they don't provide any official support for the API, we reverse engineered it from the Android application and they're under no obligations to support it, it's not like anyone is paying them any money to maintain/support it.

Snux, 2023-03-13:
I've still had no issues with this API.... no account being blocked or anything...

I don't know anything about how HA is using it, but this is what my use case is doing in case it helps...

I only use api.login once, when my python code starts.  I'll only call it again if for some reason the connection drops out.  So on the growatt side they won't see multiple login attempts.
I'm only using api.mix_detail, api.mix_totals, api.mix_system_status and api.mix_info and I call each of those every 3 minutes.

might be useful, might not be.

muppet3000, 2023-03-13:
That's pretty much exactly how the HA integration works as well. It does a login once at boot and then it hangs on to the API token for the duration.

I can't explain why some people get blocked and others don't, either way, Growatt are at liberty to do whatever they want, they're their servers after all and we're not paying anything to use them.
I'm not saying that I agree with that approach, but I can understand it. The thing that frustrates me the most is the fact they refuse to respond to any emails on the topic, so I've just given up trying!
I've started writing a new HA integration that works with grott for my own purposes and I'm hoping to have it published in the next week or two.

staticweb, 2023-03-14:
I can confirm @Snux .. I run a php modified version of this project and have changed to only login, and then pull data every minute without being blocked. Thanks for the hint @Snux

Snux, 2023-03-14:
I don't think the Growatt servers update as often as that, I seem to remember dialling mine back over a period of time and decided about 3 minutes was the quickest they'll update.

staticweb, 2023-03-14:
I just made a quick output from my database and it shows different data almost every minute..

<html>
<body>

Date | plocalload | pcharge1 | ppv1 | ppv2
-- | -- | -- | -- | --
2023-03-14 17:36:07 | 120 | 60 | 41 | 27
2023-03-14 17:35:14 | 140 | 70 | 188 | 25
2023-03-14 17:34:08 | 150 | 130 | 216 | 30
2023-03-14 17:33:08 | 150 | 130 | 216 | 30
2023-03-14 17:32:07 | 140 | 130 | 236 | 30
2023-03-14 17:31:08 | 170 | 110 | 239 | 44
2023-03-14 17:30:16 | 200 | 110 | 239 | 54
2023-03-14 17:29:06 | 200 | 130 | 245 | 73
2023-03-14 17:28:05 | 200 | 130 | 245 | 73
2023-03-14 17:27:05 | 220 | 150 | 262 | 94
2023-03-14 17:26:05 | 260 | 150 | 287 | 137
2023-03-14 17:25:14 | 240 | 240 | 217 | 197
2023-03-14 17:24:06 | 240 | 250 | 217 | 197
2023-03-14 17:23:05 | 300 | 250 | 333 | 229
2023-03-14 17:22:05 | 250 | 180 | 303 | 172
2023-03-14 17:21:06 | 240 | 130 | 250 | 131

</body>
</html>

Verminskyi, 2023-06-05:
Just added a new installation and less than 48hrs later using 300s polling the shinephone app comes up with account locked. I've added a new account which when I try to add hardware to says "Your account does not have permission to operate'.

Very frustrating especially as I'm not getting grott working let alone the implementation of that (my lack of command line Fu mostly), or the growatt data stick which I can't seem to remote in. This is a woe is me post but thanks for the hard work on trying to get the data out.