indykoning/PyPi_GrowattServer #56: No data for energy dashboard

MaciekzP, 2023-02-12:
Hello I tried to add enegy prodution in a solar power section of the dashboard, but the only entity available from scroll down menu is "Lifetime energy output".

muppet3000, 2023-02-12:
Hi there, I'm not sure what you're referring to, but I assume you're talking about the Home Assistant integration?
If so, you're in the wrong repo, you should post over here instead: 
https://github.com/muppet3000/homeassistant-growatt_server_api

MaciekzP, 2023-02-12:
OK thanks

closed 2023-02-12 by MaciekzP

muppet3000, 2023-02-12:
@MaciekzP  - I specifically recommend reading this issue too: https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15

Specifically the following posts: 
[first](https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1423787751), [second](https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1424810061), [third](https://github.com/muppet3000/homeassistant-growatt_server_api/issues/15#issuecomment-1427052222)

MaciekzP, 2023-02-12:
Addon as such works all ok. I can see all entities, it is just when I try to add production from PV I can not see correct entity.
When I try to find why, it directs me to [https://www.home-assistant.io/more-info/statistics/](url)

muppet3000, 2023-02-12:
> Addon as such works all ok. I can see all entities, it is just when I try to add production from PV I can not see correct entity. When I try to find why, it directs me to [https://www.home-assistant.io/more-info/statistics/](url)

In which case, go and log a bug over in the other repo, just bare in mind that I'm not putting a huge amount of effort on new features given the current circumstances. If there are easy bug fixes though I'll still tackle those.