indykoning/PyPi_GrowattServer #57: Server name changed?

oene, 2023-03-06:
Hi,

Wasn't able to retrieve data using server-api.growatt.com using username and password:

```
raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://server-api.growatt.com/newTwoLoginAPI.do

```

Changing the to **server.growatt.com** solved the issue for me.

```
__init__.py

class GrowattApi

    server_url = 'https://server.growatt.com/'

```

Regards,

Oene

bompaization, 2023-03-06:
picking up on this, is it possible to change the default url to this? I'm having trouble getting my build to accept the change in the module.

mreferre, 2023-05-29:
I am having the same problem. Manually hacking the init makes it work. We are waiting for [this](https://github.com/indykoning/PyPi_GrowattServer/pull/62) I assume?

marcovtwout, 2024-02-14:
Fixed by https://github.com/indykoning/PyPi_GrowattServer/pull/77
@indykoning this issue could be closed