indykoning/PyPi_GrowattServer #66: requests.exceptions.HTTPError: 403 Client Error

gickowtf, 2023-06-10:
Hi,

I try to use this nice Modul... but I ever get an error like this:

`Traceback (most recent call last):
  File "/Users/mariobach/PycharmProjects/Test/growatt_test.py", line 35, in <module>
    login_response = api.login(username, user_pass)
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/growattServer/__init__.py", line 131, in login
    response = self.session.post(self.get_url('newTwoLoginAPI.do'), data={
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/requests/sessions.py", line 637, in post
    return self.request("POST", url, data=data, json=json, **kwargs)
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/requests/sessions.py", line 710, in send
    r = dispatch_hook("response", hooks, r, **kwargs)
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/requests/hooks.py", line 30, in dispatch_hook
    _hook_data = hook(hook_data, **kwargs)
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/growattServer/__init__.py", line 41, in <lambda>
    'response': lambda response, *args, **kwargs: response.raise_for_status()
  File "/Users/mariobach/.conda/envs/untitled/lib/python3.10/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://server-api.growatt.com/newTwoLoginAPI.do`

what i do wrong?! Please Help me.

gickowtf, 2023-06-10:
I did it... its a wrong url in repo. It is Not Server-api.growatt... it is Server.growatt...

APKing1, 2023-09-30:
Since 28th September, can't login at server.growatt.com

Getting requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://server.growatt.com/newTwoLoginAPI.do

ryandrake182, 2023-09-30:
> Since 28th September, can't login at server.growatt.com
> 
> Getting requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://server.growatt.com/newTwoLoginAPI.do

Same issue as this https://github.com/indykoning/PyPi_GrowattServer/issues/74

marcovtwout, 2024-02-14:
Fixed by https://github.com/indykoning/PyPi_GrowattServer/pull/77
@indykoning this issue could be closed.

closed 2025-04-29 by indykoning