indykoning/PyPi_GrowattServer #76: [FR] Option to use official web API

Saentist, 2024-01-25:
As was suggested [by @marcovtwout ](https://github.com/indykoning/PyPi_GrowattServer/issues/74#issuecomment-1838121957)

It's better to use and official way to GET data and SET parameter's 
documentation link
https://www.showdoc.com.cn/262556420217021

Templates by device models can be made also.

And will be more informative then this basic scrape in `configuration.yaml` with work from lot of time
```yaml
multiscrape:
  - name: HA scraper
    resource: https://server.growatt.com/singlePage/getPlantData
    scan_interval: 300
    method: POST
    headers:
      Content-Type: application/x-www-form-urlencoded
    payload: 'plantId=xxxxxxx' #plant id number
    sensor:
      - unique_id: 4200TLXE_today
        name: Inv1 day 
        value_template: '{{ value_json.obj.eToday }}'
        unit_of_measurement: kWh 
      - unique_id: 4200TLXE_month
        name: Inv1 Month
        value_template: '{{ value_json.obj.eMonth }}'
        unit_of_measurement: kWh 
      - unique_id: 4200TLXE_total
        name: inv1_tot
        value_template: '{{ value_json.obj.eTotal }}'
        unit_of_measurement: kWh 
        device_class: energy
        state_class: total_increasing
      - unique_id: 4200TLXE_now
        name: inv1_realtime
        value_template: '{{ value_json.obj.pac }}'
        unit_of_measurement: kW 
```

Saentist, 2024-07-15:
End user portal
server.growatt.com

timohencken, 2025-03-28:
I'm currently trying to implement the official API in https://github.com/timohencken/GrowattPublicApiPy - but it will take a while...

Here is how you log in:

The API requires token authentication. The token can be retrieved via
* Growatt Android/iOS app "ShinePhone"
  * go to "Me" -> your username -> "API Token"
  * copy your token or request an new one
* Growatt web frontend at [openapi.growatt.com](http://openapi.growatt.com/)
  * login
  * click your username in the upper right corner
  * go to "Account Management" -> "API Token"
  * copy your token or request an new one

Token then can be passed using request header "token"