indykoning/PyPi_GrowattServer #78: Still getting error 403 with version 1.5.0

fisken, 2024-02-14:
Hi,

I just updated to 1.5.0; the previous versions last worked for me a few months ago.
I changed the URL with the previous version but had no luck.

After updating, I'm still getting:

HTTPError                                          Traceback (most recent call last)
ipython-input-4-06dc7ac3e99d in module
     10 
     11 api = growattServer.GrowattApi(False, "adgahtwq3241")
---> 12 login_response = api.login(username, user_pass)
     13 print("Add random ID to default User-Agent")
     14 print("User-Agent: %s\nLogged in User id: %s" % (api.agent_identifier, login_response['userId']))
...
Further down:
HTTPError: 403 Client Error: Forbidden for url: https://openapi.growatt.com/newTwoLoginAPI.do

I have tried with:
`api = growattServer.GrowattApi(False, "something")`
`api = growattServer.GrowattApi(True)`
`api = growattServer.GrowattApi()`

All give the same HTTPError: 403.

Am I doing something wrong? I'm a Python novice.

GraemeDBlue, 2024-02-15:
Same issue here, only found this server yesterday

Nbondb, 2024-02-19:
Same herer, running on termux on android 10.
File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/growatt/mymain", line 19, in <module>                                                login_response = api.login(username, password)                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/growatt/__init__.py", line 131, in login                                             response = self.session.post(self.get_url('newTwoLoginAPI.do'), data={                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                            File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/requests/sessions.py", line 677, in post                                             myret= self.request("POST", url, data=data, json=json, **kwargs)                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/requests/sessions.py", line 622, in request                                          resp = self.send(prep, **send_kwargs)                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^           File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/requests/sessions.py", line 752, in send                                             r = dispatch_hook("response", hooks, r, **kwargs)                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                 File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/requests/hooks.py", line 30, in dispatch_hook                                        _hook_data = hook(hook_data, **kwargs)                         ^^^^^^^^^^^^^^^^^^^^^^^^^          File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/growatt/__init__.py", line 41, in <lambda>                                           'response': lambda response, *args, **kwargs: response.raise_for_status()                                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^                         File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/requests/models.py", line 1024, in raise_for_status                                  raise HTTPError(http_error_msg, response=self)requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://openapi.growatt.com/newTwoLoginAPI.do

Nbondb, 2024-02-20:
I don't know if relevant but I couldn't run setup.py because it couldn't find setuptools

GraemeDBlue, 2024-02-20:
I've tried https://github.com/PLCHome/growatt and this works fine so will see if I can update they  python to work the same

Nbondb, 2024-02-21:
Now managed to run setup.py but no difference.
I'm inexperienced in python and http, but added prints into the code and notice that there is no Authorization element in the http header, is that the problem?

Nbondb, 2024-03-12:
Is it feasible to convert it to Python? Is it likely to take a long time?
Many thanks for your efforts.

> I've tried https://github.com/PLCHome/growatt and this works fine so will see if I can update they python to work the same

GraemeDBlue, 2024-03-12:
> Is it feasible to convert it to Python? Is it likely to take a long time? Many thanks for your efforts.
> 
> > I've tried https://github.com/PLCHome/growatt and this works fine so will see if I can update they python to work the same

Its a lot of work but I will see what I can do, Python not my daily language so will be a bit of a learning curve for me first

sledgemhammer, 2024-03-14:
you have to change <agent_identifier => value  to something real the default in the example is blacklisted

ericsfraga, 2024-03-29:
I also am having problems connecting. If I use one server (`openapi-us.growatt.com`), I get `Username or Password Error`. If I use `server.growatt.com`, I get `403 Client Error: Forbidden for url [...]`.

This is with `True` as the argument to the `GrowattApi` function. If I need to specify an actual agent identifier, can somebody suggest what I should use?

For context, I used the 1.2 version of this package for some time but it quit working a long time ago now. I can access the data via the web or the Android app so I know my username/password are okay.

Thank you.

sledgemhammer, 2024-03-29:
> I also am having problems connecting. If I use one server (`openapi-us.growatt.com`), I get `Username or Password Error`. If I use `server.growatt.com`, I get `403 Client Error: Forbidden for url [...]`.
> 
> This is with `True` as the argument to the `GrowattApi` function. If I need to specify an actual agent identifier, can somebody suggest what I should use?
> 
> For context, I used the 1.2 version of this package for some time but it quit working a long time ago now. I can access the data via the web or the Android app so I know my username/password are okay.
> 
> Thank you.
> 
> **Update**: the JavaScript code referred to above works just fine. I can use that for what I need but a python script would be better overall.

Did you change the agent identifier in the pyhton script to a real current chrome or other real agent string ? I had exactly the same until i changed that. Note that the servers in use are bound to global regions os you cannot random change  that. Pyhton running flawles here every x minutes after that change.

sledgemhammer, 2024-03-29:
Had the same u til i changed the agent.

flopp999, 2024-03-29:
https://github.com/indykoning/PyPi_GrowattServer/blob/master/growattServer/__init__.py#L28
Change to "Chrome" is working for me

flopp999, 2024-03-29:
Or use below when initialise to use a random agentID. Below is copied from Readme
api = growattServer.GrowattApi(True)

ericsfraga, 2024-04-03:
Thank you. I did try the random ID case (i.e. with True as an argument) but that did not help. I haven't tried actually specifying a string for the agent. The javascript code is working, as I indicated in my update to the post, and that's good enough for me for the moment.

molekuul, 2024-06-14:
I have the same problem, I also get a similar error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/growattServer/__init__.py", line 131, in login
    response = self.session.post(self.get_url('newTwoLoginAPI.do'), data={
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/requests/sessions.py", line 637, in post
    return self.request("POST", url, data=data, json=json, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/requests/sessions.py", line 710, in send
    r = dispatch_hook("response", hooks, r, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/requests/hooks.py", line 30, in dispatch_hook
    _hook_data = hook(hook_data, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/growattServer/__init__.py", line 41, in <lambda>
    'response': lambda response, *args, **kwargs: response.raise_for_status()
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jorisweijters/.pyenv/versions/3.11.3/lib/python3.11/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://openapi.growatt.com/newTwoLoginAPI.do
>>>
```

please help,

Kind regards,
Joris Weijters

molekuul, 2024-06-15:
@sledgemhammer Thanks for you comments, I can get further with this.