indykoning/PyPi_GrowattServer #81: Set Export Limit

Ashutor, 2024-05-12:
Is it possible to fetch and control the Export Limit (on / off) with the current functionality? - or optionally request this feature to be added?

I have been unable to identify the correct setting if it is already available.

Tampa, 2024-05-15:
Looking through the settings I haven't spotted it either, but it probably is there. Probably just needs the correct call added to it. Far as I can tell all the website does is talk to that api as well, so might just be the field name even. I want to set this too, to make it stop exporting when battery charge drops below 50%, which weirdly you can't set otherwise in a load first setup.

(I dread to think what happens when all this cloud stuff goes away, hopefully they do handover ways to clone the api)

Waester, 2025-04-12:
For anyone else looking for this information later, "backflow_setting" is the setting corresponding to "Export Limit (on / off)". Accept value ranging from 0-2.