indykoning/PyPi_GrowattServer #83: Is the userId always the same?

simonf2004, 2024-06-27:
Hello, i have started using this library and i am getting the 403 error about 40% of time i try to login (even though i have changed the user agent). 
I think one possible solution to this might be to login just once, keep the userId (that the login returns) and just get the data with this userid. 
My question is: Is userId always the same or does it change from time to time. If it was always the same i could just login once and then always use the obtained userid no?

Thank you very much and have a great day

indykoning, 2024-07-03:
I'm pretty sure the userId is indeed always the same, however i do not believe the userId is authentication. Rather i'm guessing a cookie or session is started and thus you need to log in before anything else. 

You could try it though, by logging in using one script. Then starting another script to get the plant list using that userId

simonf2004, 2024-07-04:
Yes, you are right. The Userid seems to be always the same and it isnt authenticiation. It wouldnt really make sense. Thanks for your reply

closed 2025-05-11 by indykoning