indykoning/PyPi_GrowattServer #88: noah auslesen funktioniert nicht

liv-in-sky, 2024-09-29:
@Rappelkiste98 
@indykoning 
hi - wollte meinen noah 2000 auslesen mit dem beispiel script - leider bekomme ich einen fehler in zeile 67 [noah_example.py](https://github.com/indykoning/PyPi_GrowattServer/blob/master/examples/noah_example.py)

der fehler sieht folgend aus:

`
python3 noahsample.py
Enter username:xxxxxx
Enter password:
{   'data': [   {   'currentPower': '0 W',
                    'isHaveStorage': 'false',
                    'plantId': '2222207',
                    'plantMoneyText': '0.0 ',
                    'plantName': 'MySolar',
                    'todayEnergy': '0.6 kWh',
                    'totalEnergy': '101.5 kWh'}],
    'success': True,
    'totalData': {   'CO2Sum': '40.6 T',
                     'currentPowerSum': '0 W',
                     'eTotalMoneyText': '0.0 ',
                     'isHaveStorage': 'false',
                     'todayEnergySum': '0.6 kWh',
                     'totalEnergySum': '101.5 kWh'}}
***Totals for all plants***

***List of plants***
  ID: 2222207, Name: MySolar

***Info for Plant 2222207- MySolar***
  CO2 Reducion: 108.12
  Nominal Power (w): 1600000
  Solar Energy Today (kw): 1.6
  Solar Energy Total (kw): 270.3

  Devices in plant:
    - Device - SN: BZP2N44444, Type: tlx
Traceback (most recent call last):
  File "/root/noahsample.py", line 67, in <module>
    is_noah = api.is_plant_noah_system(plant['plantId'])
              ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'GrowattApi' object has no attribute 'is_plant_noah_system'

----------------------

könntet ihr mir bitte  dabei helfen - ich versuche ein script zu bekommen, welches nur die leistungsabgabe steuert (Verbraucher zuerst) steuern kann - dabei habe ich gehofft, dieses script zu nutzen - leider kenn ich phyton nicht und ich stolpere schon bei dem example script :-(

`

xMichl, 2024-10-01:
I have exactly the same problem / response from the noah_example script. Already checked the __init__.py, but could't find a solution. Please help @indykoning

liv-in-sky, 2024-10-02:
@xMichl 

hi, danke, das du das teilst - ich würde gerne deine lösung anwenden - ich kopiere das init_py file von der github seite hier und speichere es in meinem server unter /usr/local/lib/python3.11/dist-packages/growattServer

das funktioniert leider so nicht - hast du einen tipp für mich, wie das genau funktioniert

liv-in-sky, 2024-10-02:
@xMichl 

hast du mal die set befehle ausprobiert und evtl ein script beispiel

z.b. Change "System Default Output Power"   **oder**

Change "Operation Mode" Time Segment"

kenn mich mit phyton leider garnicht aus :-(

xMichl, 2024-10-02:
> @xMichl
> 
> hi, danke, das du das teilst - ich würde gerne deine lösung anwenden - ich kopiere das init_py file von der github seite hier und speichere es in meinem server unter /usr/local/lib/python3.11/dist-packages/growattServer
> 
> das funktioniert leider so nicht - hast du einen tipp für mich, wie das genau funktioniert
> 
> **EDIT:** - hat sich erledigt - auslesen funktioniert jetzt auch bei mir

Wäre auch gut, wenn du deine Lösung dann für andere teilst. Ich hehme an du hast den flaschen Pfad genommen. Die init.py befindet sich zumindest bei mir in /home/pi/.local/lib/python3.x/site-packages/growattServer

xMichl, 2024-10-02:
> @xMichl
> 
> hast du mal die set befehle ausprobiert und evtl ein script beispiel
> 
> z.b. Change "System Default Output Power" **oder**
> 
> Change "Operation Mode" Time Segment"
> 
> kenn mich mit phyton leider garnicht aus :-(

Ich kenn mich auch nicht aus. Kann es nur grob lesen und versuche es nachvollziehen. Bin dran was anhand der example-files zu basteln. Melde mich wenn es klappt.

liv-in-sky, 2024-10-02:
sorry - ich habe das file wie oben im pfad (nutze proxmox lxc) /usr/local/lib/python3.11/dist-packages/growattServer kopiert
(habe im server einfach nach growattServer gesucht find / -name "growattServer"

ich habe zum kopieren den wget befehl genommen und der hat nicht das richtige file kopiert - also dein weg funktioniert

liv-in-sky, 2024-10-02:
habe im [noah_example.py](https://github.com/indykoning/PyPi_GrowattServer/blob/master/examples/noah_example.py) einfach mal  folgendes eingefügt response = api.update_noah_settings('default_power',[350])
- dann kommt:

`....

  Remaining battery (%): 12
  Solar Power (w): 30
  Charge Power (w): 0
  Discharge Power (w): 0
  Output Power (w): 0
Traceback (most recent call last):
  File "/root/noahsample.py", line 92, in <module>
    response = api.update_noah_settings('default_power',[350])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: GrowattApi.update_noah_settings() missing 1 required positional argument: 'parameters'`

liv-in-sky, 2024-10-02:
@xMichl 

da fehlt was in der befehlsbeschreibung - die sereinnummer des noah

so geht das setting :-)

`response = api.update_noah_settings('0PVP20ZRxxxxxx36','default_power',['150'])`

xMichl, 2024-10-02:
Jo läuft, hab ein kleines Script draus gebastelt, das einen bestimmten Wert einstellt: 

_import growattServer
import datetime
import getpass
"""
Script to change default_output_power on noah system to 150W
"""
api = growattServer.GrowattApi(False, "Chrome")
login_response = api.login('Username',
                           'Password')
api.update_noah_settings('serial_noah', 'default_power', ['150'])_

Username, Password und serial_noah müssen entsprechend angepasst werden.

Habe hiervon pro 50W (beginnend ab 150W bis 800W) jeweils ein Script erstellt. Habe vor, dieses dann via iobroker javascript adapter jeweils zu triggern, wenn sich der Datenpunkt bzgl. Stromverbrauch, der über einen Hitchilesekopf und den Smartmeter-Adapter generiert wird, in einen entsprechenden Bereich ändert. 

Besser wäre es natürlich ein Script zu erstellen, das bspw. alle 15 Sekunden von Smartmeteradapter den Wert abgreift und dann basierend darauf automatisch den entsprechenden Wert an den Noah sendet. Habe aber null Ahnung, wie und wo man das im iobroker programmiert. Aber ich denke die 50W Schritte sollten so auch gut ausreichen... Auch wenn es etwas umständlich ist.

liv-in-sky, 2024-10-02:
mit diesem script kannst du den gewünschten wert direkt beim aufruf eingeben

`import growattServer
import datetime
import getpass
import sys

"""
Script to change default_output_power on noah system to 150W
"""
api = growattServer.GrowattApi(False, "Chrome")
login_response = api.login('user',
'passw')
api.update_noah_settings(serialnuberNoah', 'default_power', [sys.argv[1]])`

beispiel
python3 scriptname.py 150

ml1969, 2025-03-07:
Hallo,

stehe gerade auch auf dem Schlauch.
Habe das obige Script-Beispiel verwendet und bekomme:

_AttributeError: 'GrowattApi' object has no attribute 'update_noah_settings'. Did you mean: 'update_plant_settings'?_

komme nicht weiter.....

MIchael

closed 2025-05-11 by indykoning